RobSmithDev/ArduinoFloppyDiskReader #1: Question-WD1793FDC compatible use?

drencorxeen, 2018-02-05:
This is not really a issue, but more of a question.

With the code you have here for working with Amiga disks.  How hard would it be to modify the code to work with floppy disks that were formatted by WD1793FDC's?

The reason I ask is that I am trying to find a way to read and write to old Tandy/TRS-80 Color Computer floppy disks.  This includes both the 250Kbps mode of 5.25"/3.5" and the 8" 500Kbps drives.  The default sector size is 256 Bytes.

Thanks for any information you might have on this crazy question.

RobSmithDev, 2018-02-05:
Hi yes it’s possible I guess.  
Without checking I would imagine the 500kbps is MFM encoded and the 250kbps being FM encoded.  The software will decide the MFM format, and FM is really simple

The Arduino code is basically sending pulse data back to the computer so could read these disks, however you would have to change the timer/math calculations to take into account the different data rate (off the top of my head it’s currently set for 300kbps)

drencorxeen, 2018-02-06:
Thanks for the information.

Yeah the 250Kbps and 500Kbps are both MFM in this case.

closed 2018-03-28 by RobSmithDev