RobSmithDev/ArduinoFloppyDiskReader #5: Writing doesn't work for me

maps82, 2018-11-03:
Unfortunately I never managed to get floppy disk writing to work. I always get: "An unknown response was received from the Arduino while executing the WriteTrack command". I would be glad if some could help me here. I tried to contact Rob via hackster.io but without response.

I tried multiple PCs, FTDI boards and Floppy Disk Drives.

Original message on hackster was:

"Hello Robert!

I built your Amiga Floppy Disk Reader/Writer. First thank you very much for your great work engineering it :)

Unfortunately I cannot write to disks. Every else seems to work but I can't actually ready disks w/o checksum errors as I don't have a single Amiga Floppy - I built the thing to write them.

The Diagnostics runs fine until it comes to writing. Then I get the message "An unknown response was received from the Arduino while executing the WriteTrack command". Same message when I try to write an ADF to the Disk (no matter if erase or verify is on or off).

I read that it often has to do with the CTS line, but I checked it multiple times and the connection from the FTDI to the Arduino is solid. What did work: I once connected the CTS line to Arduino A3 instead of A2 (A3 is unused) and adjusted the sketch accordingly. Strange thing: In diagnostics mode I got a CTS error, but writing without verifying worked for the first time! But the disk was not readable on the Amiga.

Do you have any tips for me to get it working? I spent hours finding the error. Should I try other FTDI boards? Here you can see pictures of the current board and setup:

https://ibb.co/hYszUz
https://ibb.co/ngP2hK

Thank you very much!

PS: I tried different USB ports (without HUB), Floppy is a Mitsumi, Cable is 10-15cm, I use an additional 5V power supply, voltage is stable also when the motor starts spinning."

RobSmithDev, 2018-11-03:
Hi

Thanks for getting in contact

The only time this happens seems to be with some of the FTDI boards.  I’m not sure which.  Some work perfectly and some don’t.  It’s something to do with the timing of the CTS lines. I don't know if it’s genuine vs fake. 

I know that doesn’t really help much but that’s the cause of the issue anyway.

Best wishes

Rob Smith
Creative/Frontend/Backend Developer
http://robsmithdev.co.uk

maps82, 2018-11-03:
Thanks for your response! Can you recommend a specific FTDI board that works?

maps82, 2018-11-04:
I ordered a third FTDI board on eBay, which looks different than the ones I have. Will keep you updated.

maps82, 2018-11-13:
I got the new FTDI board and although it looks exactly as my previous one I successfully wrote a Workbench disk today which could be booted by the Amiga :)

Unfortunately verifying doesn't work (I have to disable it) and I have no success with Pinball Dreams and another Demo.

When it comes to writing in Diagonistics mode I don't get anymore the message:

> An unknown response was received from the Arduino while executing the WriteTrack command

but

> Writing and Verifying Track 41 (Upper Side).
DIAGNOSTICS FAILED: Unable to detect written track.  This could be for one of the following reasons:

RobSmithDev, 2018-11-15:
Hi

That’s weird - still sounds a little dodgy if it won’t validate and you get be same error but hey

Best wishes

Rob Smith
Creative/Frontend/Backend Developer
http://robsmithdev.co.uk

jtsiomb, 2018-12-20:
> Is every disk supposed to work (Games, Demos, NDOS...) or only Amiga Dos formatted ones?

Every disk is supposed to work. I've written (with my AVR version of Rob's design, and my port of his firmware to straight AVR) every kind of disk. Including multiple demos and games. In fact this is my primary means of writting amiga disks now, since it's much easier than writing with the amiga itself through the serial port.

Since your errors seem to be intermittent, I would suggest trying to wire up a more permanent version on perfboard, soldered together, instead of the breadboard you're showing in the pictures. Maybe some connection is dodgy. An alternative suggestion is to try making my version on perfboard or PCB. I include a schematic which you could use for wiring it up manually on a perfboard, and an untested PCB design derived from it (untested because I made mine before making the PCB design, and I never bothered to make another one). Unfortunately this would require soldering an FTDI chip, which is never fun, but you could find a through-hole breakout instead. https://github.com/jtsiomb/usbamigafloppy

codmpm, 2019-02-04:
@gianlucarenzi would you mind open-source the pcb layout?

gianlucarenzi, 2019-02-05:
On 02/05/2019 12:00 AM, Patrik Mayer wrote:
> @gianlucarenzi <https://github.com/gianlucarenzi> would you mind
> open-source the pcb layout?
>

yes

-- 
Eurek s.r.l.                          |
Electronic Engineering                | http://www.eurek.it
via Celletta 8/B, 40026 Imola, Italy  | Phone: +39-(0)542-609120
p.iva 00690621206 - c.f. 04020030377  | Fax:   +39-(0)542-609212

prickle, 2020-11-11:
I had this issue too, but with CH340G as well as FTDI. After fiddling with hardware for days I seem to have fixed it by changing the flow control code in CHECK_SERIAL().

```
#define CHECK_SERIAL()          if (UCSR0A & ( 1 << RXC0 )) {                      \
                                    SERIAL_BUFFER[serialWritePos++] = UDR0;        \
                                    serialBytesInUse++;                            \
                                }                                                  \
                                if (serialBytesInUse<SERIAL_BUFFER_START)          \
                                    PIN_CTS_PORT &= (~PIN_CTS_MASK);               \
                                else PIN_CTS_PORT|=PIN_CTS_MASK;                   \
                                
```

RobSmithDev, 2020-11-11:
Hi

Interesting.  I would recommend this as the pc could write data faster than it’s written to the disk.  That being said, it does suggest that some of the FTDI/CH340G chips are probably a bit slower in the CTS line so this could explain a few things.

I’ll make a note to go back and investigate this further

prickle, 2020-11-24:
Just to confirm, with the bugfix given above I have successfully read and written disks that can be read by a real Amiga. Thanks Rob, this is a great project! Here's my finished unit. This drive seems to run fine from USB power only.
[image]

RobSmithDev, 2020-11-24:
Thanks, I’ll give this a test with previous working hardware and see if it still works with them

Rob Smith
Creative/Frontend/Backend Developer
http://robsmithdev.co.uk

RobSmithDev, 2020-11-24:
Just testing this CHECK_SERIAL change out with my "working" drive.  All looks good.  I was a little concerned that it might mean data might get written to fast, but I guess it seems fine.  Just written and verified several disks with no issue.

It'll be included in the next release

closed 2020-11-24 by RobSmithDev