RobSmithDev/ArduinoFloppyDiskReader #7: Ported the sketch to work with Arduino Pro Micro (Atmega32U4, 5V, 16Mhz)

hpingel, 2019-08-05:
Hi Robert,

Thank you for your work on this and sharing the project as open source software! Greatly appreciated! I built your project last weekend on a breadboard. I only had an Arduino Pro Micro at hand + a  FTDI thingy. As the Pro Micro differs from the Mini I wasn't sure if I could make it work.

Long story short: By changing some defines in your sketch I am able to run the software on an Arduino Pro Micro 5V 16Mhz. The Pro Micro has it's own USB port but this is currently unused as the communication to the PC goes via FTDI.

If you are interested I can share my code changes. I was wondering if there would be any way to use the Pro Micro without the FTDI dongle in future. But I don't know where the CTS pin would end up.

Cheers,
Henning

biotech80, 2020-07-06:
Hi,

I'm going to make some of them using Arduino Pro Micro 5V 16Mhz.
What changes does it need?

I see they share the same pinout.

Thank you!

locka99, 2021-01-02:
I'm a little confused as to when FTDI is required and when it isn't. I have a Mega 2560 so I was hoping to give it a try although that also has a USB. I'm of the impression that the onboard USB serial port isn't fast enough for the baud but it wasn't clear to me why that would be the case or how adding an FTDI to a board which already has USB makes a difference. Or if people who only need read-only can scrape by without the FTDI.

I'm ordering an FTDI board & Pro anyway ($2 clones) but perhaps if the onboard USB is good enough, especially for read-only then perhaps the docs could be more clear on the reasons for the FTDI and situations where it isn't required.

RobSmithDev, 2021-01-02:
Hi

The separate board was required for writing disks only. The arduino was having hardware flow control (CTX) pin on the serial port to control the flow of data from the Pc, but that pin isn’t exposed on the onboard serial devices. So for reading, yes the onboard one is fine

Hope that makes sense

Rob Smith
Creative/Frontend/Backend Developer
http://robsmithdev.co.uk

locka99, 2021-01-02:
Thanks for the clarification I think I understand now. I'll try with the Mega 2560 I have report any issues I have.

closed 2021-01-27 by RobSmithDev