RobSmithDev/ArduinoFloppyDiskReader #14: Write errors in first sectors

mlorenzati, 2021-01-05:
On a PCB I built (which I can share the schematic and pcb) (no additional ftdi if using a CH340 based nano clone) I see that event though test are passing and write shows success (on a write and verify scenario) when reading the written disk, first sector tend to be not properly written.
Wiring and power are ok, FFD and disk are not a known good since I dont have other systems to try it.

BTW I tested with the pre built windows executable.

RobSmithDev, 2021-01-27:
Out of interest, can you share a link to this CH340 nano clone? Im curious to see how CTS is accessed

mlorenzati, 2021-01-28:
Sure, have a look here:
https://cdn.hackaday.io/files/293271254726368/nano_ch340_schematics-rev1.pdf

I added an extra pin to the nano and wired it to pin 9 of ch340
Check that pin 9 (CTS) is near a border of the nano PCB, you can leverage the hole the pcb has there to place the additional pin.

Since as you said arduino nano is just to tight for timing, I moved to Greaseweazel project which uses a STM32F103 Cortex M3 at 72Mhz

RobSmithDev, 2021-05-28:
Discovered that CH340 chips are nasty, so closing this.  Please switch to FTDI chips.