RobSmithDev/ArduinoFloppyDiskReader #15: Checksum errors while reading

axilos, 2021-01-15:
I found two old Amiga diskettes from 1996 :) I created two games at that time :)
So I wanted to see how the games work and came up at this solution with Arduino.
After ordering all componets and soldering wires I had problem with power supply for the floppy drive and decided to replace it with cell phone charger where I have measured 5.12V.
Diagnostics went fine. At least reading. I didn't test writting because I don't want to destroy my floppies and I don't have any extra.
After clicking on Copy Disk the floppy drive LED turn on but everything stops at Current Track: 1 Current Side: Lower with error message: Disk has checksum errors/missing/damaged data.
After making few minutes break and tried again it stops at track 2 lower.
Clicking on Retry do not help.
I tried both diskettes and the problem is the same.
Any ideas what can I try to read those diskettes?

RobSmithDev, 2021-01-15:
Hi

I suspect three possible reasons:

1. The usb power supply isn’t strong enough 
2. The drive heads are dirty
3. The disks have collected a little mould on them (it’s not uncommon) - you’ll have to slide the gutter back and check

I did have one drive that did this and I think the head was mis-aligned.

Hope that helps

Rob Smith
Creative/Frontend/Backend Developer
http://robsmithdev.co.uk

axilos, 2021-01-16:
Hi Rob

thanks for very fast reply.

I checked [datasheet](https://datasheetspdf.com/pdf-file/900360/Samsung/SFD-321B/1) of floppy drive. Reading operation consume 0.3 A. Huawei phone charger has 2A. I sounds enough.
Diskettes are very clean. They were closed inside the plastic bag.
I don't know how to check drive heads. I will order new floppy drive and try again.

Can you implement 'Continue' button? But not 'Retry' button like now.
Explanation: If I click 'Copy Disk' button it reads up to 5 tracks and stuck. After poping up checksum error message I would like to click Abort so that floppy drive is OFF and than under 'Copy Disk' button to have 'Continue' button so it start reading from last read track?

RobSmithDev, 2021-01-16:
Hi

There should be an ignore option, although I cant remember if it appears the first time or after a few. Are you running the console or window program?

Rob Smith
Creative/Frontend/Backend Developer
http://robsmithdev.co.uk

axilos, 2021-01-16:
I'm running window program. I will try console now.
I started to take I look on ArduinoFloppyReaderWinDlg.cpp. Maybe I can add 'Continue' button. There are some errors like afx***.h files are missing.
'Ignore' button keeps floppy drive ON.

It depands, sometimes it runs up to track 5 in one shot, those reading should be fine. But sometimes it reads only 1 or 2 tracks. I just wanted to start from 5 and combine older with newer read. I don't want to start from 0 track every time.

axilos, 2021-01-16:
With console version is the same. Usually stuck at track 1 lower side. Sometimes it goes to track 2.

kollokollo, 2021-01-16:
This can happen with very old disks. I have read some of the tracks up to 20 times and then collect the good sectors together to get a final result.

RobSmithDev, 2021-01-16:
If you leave it long enough you should get the option to ignore, but you can always modify the code to do this differently. 

The longer it runs the more dodgy data it collects which it may be able to recreate the original sector from 

Rob Smith
Creative/Frontend/Backend Developer
http://robsmithdev.co.uk

axilos, 2021-01-16:
Popup window with Abort, Retry, Ignore buttons is shown pretty fast. I tried to click Ignore button but I had to do that 1-3 times per track per side. I think only track 5 went well without Ignore button.
I have ordered two floppy drives and I will see in 3 days when they arrive is there any improvement.

axilos, 2021-01-27:
I can't convert amiga floppies to ADF files and I will throw all 3 floppy drives away. Gave my best... I will hold my floppies alive and wait better future. Maybe one day I could send to someone who can convert them.

RobSmithDev, 2021-01-27:
Before you throw them out, hold on, I have an update coming soon, and whilst its not going to fix everything it does improve reading of some disks.

axilos, 2021-01-27:
OK, I will get them out of trash can. Please write here when update is on. I wish I can help you somehow.

RobSmithDev, 2021-05-28:
Ok, new versions are all available now.  Closing this so if there are further issues you can open a new one.

closed 2021-05-28 by RobSmithDev