RobSmithDev/ArduinoFloppyDiskReader #16: Porting to RaspberryPi

frimdo, 2021-03-11:
Hi, I would like to implement this feature in RaspberryPi. I am thinking, that I would port the arduino part to bare c/c++ and use linux scripts. My questions are: 
- didn't somebody already do that? 
- Is there something I should pay extra attention to? 
- Would you suggest some particular steps?

BTW this project is awesome. Thank you for great comments and documentation.

kollokollo, 2021-03-11:
Hm. If you want to use Raspbian, this could be tricky, because Raspbian is not a real-time operating system. It would make more sense to only port the PC-part to raspbian and connect an arduino (nano) to the RaspBerry Pi for the time critical part. Basically thats what i did. I have written my own scripts for data evaluation. But having a nice GUI would of course be preferrable.

frimdo, 2021-03-11:
Oh, that is exactly what I am asking for. You are right. I planned to use tinyCore, but the issue with realtime is the same. Ok, I will go with the way of separating arduino part. Thanks!

closed 2021-03-11 by frimdo