RobSmithDev/ArduinoFloppyDiskReader #17: It does not run under WindowsXP

sblendorio, 2021-03-15:
Launching the EXE file, this dialog is raised:
[image]

("it's not a valid Win32 application")

RobSmithDev, 2021-03-15:
Hi

It wasn’t compiled to run under windows XP. If you want to run it on there you will have to re-compile it yourself.  The source code is available to do so

Best wishes

Rob Smith
Creative/Frontend/Backend Developer
http://robsmithdev.co.uk

sblendorio, 2021-03-15:
Which compiler should I use under Windows XP?

Thanks in advance

sblendorio, 2021-03-15:
Using Visual Studio Community with proper build tools

sblendorio, 2021-03-15:
Here is the EXE compatible with Windows XP: https://www.dropbox.com/s/kp4kt5jmccn946o/ArduinoFloppyReaderWinXP.zip?dl=0

closed 2021-03-20 by sblendorio