RobSmithDev/ArduinoFloppyDiskReader #19: Broken access on Linux

sblendorio, 2021-07-23:
Using command-line **ArduinoReaderWriter** to read a fully-functional floppy (I tried with several floppies, successfully dumped on Windows), the process does not even start, with an error of **Disk has checksum errors/mssing data.**

I tried several Linux machines and several floppies, with the same result.

Here is a video:
[image]

gianlucarenzi, 2021-07-23:
Do the same here. Debugging the
`ADFResult ADFWriter::DiskToADF(const std::wstring& outputFile, const unsigned int numTracks, std::function < WriteResponse(const int currentTrack, const DiskSurface currentSide, const int retryCounter, const int sectorsFound, const int badSectorsFound)> callback) {`
function, I can see the m_device.readCurrentTrack(data, false) returns DiagnosticResponse::drOk, but no valid sector has found. So the error:

> $ ./ArduinoReaderWriter /dev/ttyUSB0 TinyBubble.adf READ
> Arduino Amiga ADF & SCP Floppy Disk Reader/Writer V2.6, Copyright (C) 2017-2021 Robert Smith
> Full sourcecode and documentation at https://amiga.robsmithdev.co.uk
> This is free software licenced under the GNU General Public Licence V3
>
>
>Create ADF from disk mode
>
>Disk has checksum errors/missing data.  [R]etry, [I]gnore, [A]bort?

It retries for 20 times searching for sector 0/11 and it quits like above.

The SerialIO::read() looks fine to me... 

I am running Debian Buster 10.10 on a X86_64 machine. 

On the same machine on VirtualBox and the GUI Executable for Windows works for reading.

Here is the DIAGNOSTIC output on Linux:

> $ ./ArduinoReaderWriter /dev/ttyUSB0 DIAGNOSTIC
>Arduino Amiga ADF & SCP Floppy Disk Reader/Writer V2.6, Copyright (C) 2017-2021 Robert Smith
>Full sourcecode and documentation at https://amiga.robsmithdev.co.uk
>This is free software licenced under the GNU General Public Licence V3
>
> Running diagnostics on COM port: /dev/ttyUSB0
>
> Please insert a *write protected* disk in the drive.
>
> Use a disk that you don't mind being erased.
>
>This disk must contain data/formatted as an AmigaDOS disk [Enter/ESC]: 
>
>Attempting to open and use /dev/ttyUSB0 without CTS
>Testing CTS pin
>CTS OK.  Reconnecting with CTS enabled
>Board is running firmware version 1.8 (modded for DiskChange support)
>
>Testing USB->Serial transfer speed (read)
>Read speed test passed.  USB to serial converter is functioning correctly!
>Testing write-protect signal
>Enabling the drive (please listen and watch the drive)
>Did the floppy disk start spinning, and did the drive LED switch on? [Y/N]: Y
>Asking the Arduino to find Track 0
>Track 0 was found.  Asking the Arduino to find Track 70
>Could you hear the head moving quite a distance? [Y/N]: Y
>Testing Disk Change pin.
>*** Please remove disk from drive *** (you have 30 seconds)
>*** Please re-insert disk into drive *** (you have 30 seconds)
>Disk change is working correctly
>Starting drive, and seeking to track 40.
>Checking for INDEX pulse from drive
>Checking for DATA from drive
>Attempting to read a track from the UPPER side of the disk
>Attempting to read a track from the LOWER side of the disk
>Reading was successful!
>Would you like to test writing to a disk?  Please insert a WRITE ENABLED disk that you *do not mind* being overwritten [Y/N]:Y
>Writing and Verifying Track 41 (Upper Side).
>DIAGNOSTICS FAILED: Unable to detect written track.  This could be for one of the following reasons:
>DIAGNOSTICS FAILED: 1.  Please check the following PINS on the Arduino: 3, A0, A1
>DIAGNOSTICS FAILED: 2.  Please check the Arduino IDE config has not been modified from stock.  This was tested using 1.8.4, compiler settings may affect performance
>DIAGNOSTICS FAILED: 3.  Check for poor connections, typically on a breadboard they may be intermittent which may pass the above results but still not work.
>DIAGNOSTICS FAILED: 4.  Check for an electrically noisy environment.  It is possible that electronic noise (eg: from a cell phone) may cause errors reading and writing to the disk
>DIAGNOSTICS FAILED: 5.  Shorten the floppy disk cable to help reduce noise.
>DIAGNOSTICS FAILED: 6.  Ensure your power supply is strong enough to power the floppy drive.  This drive may need too much power for the USB port
>DIAGNOSTICS FAILED: 7.  You can contact me for help, but some basic electronics diagnostics will help, checkout YouTube for guides.

Running the DIAGNOSTIC on the same machine in a VirtualBox Environment it gives me the following screen:

[image]

RobSmithDev, 2021-07-23:
I think we know what this is. If you head over to the discord server and checkout the changes people are trying for Amiberry support I suspect they will fix this too

gianlucarenzi, 2021-07-23:
May I have the discord link please?

RobSmithDev, 2021-07-23:
Hi

It’s https://discord.gg/HctVgSFEXu

Do a search for VMIN there’s two lines in code and a Linux config change to make

Rob Smith
Creative/Frontend/Backend Developer
http://robsmithdev.co.uk