RobSmithDev/ArduinoFloppyDiskReader #25: BBC Micro - floppy drive

patvdleer, 2022-11-11:
Any clue if this would work for a 5.1/4 floppy drive for the BBC Micro? I initially focused on getting the machine back up and running, since it now boots I want to backup/dump the floppies first. The floppy drive I have has a card edge style 34 pin connector with MOLEX (not MOLEX Berg) for power.

RobSmithDev, 2022-11-12:
Hi, tbh I’m not sure, I suspect not directly. I suspect the code would need modifying for a different bit-rate

patvdleer, 2022-11-13:
I had some old Arduino Mini pro and a (possibly even older) FTDI in the closet, after flashing it with the ino I can't seem to be able to address it all. I've compiled the ArduinoFloppyReader drawbridge but it doesn't seem to be able to communicate with the Arduino. I'll order a new Arduino and FTDI tomorrow and see what happens.

patvdleer, 2022-11-14:
It works right away when I added `sudo`.... no clue why since I managed to flash without

patvdleer, 2022-11-14:
Is it ok to have the 1K ohm resistor between 5v of the arduino and pin 8 with separate power sources or should it be connected to the PSU of the floppy drive?

http://amiga.robsmithdev.co.uk/images/promini_writer_plus.gif

The first attempt is with the 5v not connected to the arduino, second is with it connected. I think the arduino is getting it's power over the IDC 34pin since the Arduino does turn on when only the floppy is hooked up and powered so without the FTDI connected.

```sh
patrick@patrick-X470-AORUS-ULTRA-GAMING:~/git/ArduinoFloppyDiskReader/ArduinoFloppyReader/ArduinoFloppyReader$ sudo ./drawbridge /dev/ttyUSB0 OutputFilename.SCP READ
DrawBridge aka Arduino Floppy Disk Reader/Writer V2.8.8, Copyright (C) 2017-2022 Robert Smith
Full sourcecode and documentation at https://amiga.robsmithdev.co.uk
This is free software licenced under the GNU General Public Licence V3

Error opening COM port: Error reading response from the Arduino while running command EnableMotor.  

patrick@patrick-X470-AORUS-ULTRA-GAMING:~/git/ArduinoFloppyDiskReader/ArduinoFloppyReader/ArduinoFloppyReader$ sudo ./drawbridge /dev/ttyUSB0 OutputFilename.SCP READ
DrawBridge aka Arduino Floppy Disk Reader/Writer V2.8.8, Copyright (C) 2017-2022 Robert Smith
Full sourcecode and documentation at https://amiga.robsmithdev.co.uk
This is free software licenced under the GNU General Public Licence V3

Error opening COM port: Last command completed successfully.  
```

RobSmithDev, 2022-11-14:
Hi, first you always need to power up the Arduino first as (as you suspect) it will power up from the 34-pin connector, and not always in a good way.

Re the resistor. It doesnt matter as long as its pulled to 5v.  Probably the Arduino 5V is better.

patvdleer, 2022-11-14:
I butchered the wiring and hooked up the pin 8 resistor to an additional lead from the PSU but still the same result.

As soon as I try to read from floppy the PSU shuts off, the PSU I'm using is an older ATX power supply with a short in the 20/24 pin header to keep it on

patvdleer, 2022-11-14:
I am, it's the ATX PSU that shuts off, both with and without the 5V from the FTDI hooked up to the Arduino.

RobSmithDev, 2022-11-14:
The ATX power supply shuts off?

RobSmithDev, 2022-11-14:
Looking around, and I don't know which model of BBC drive you have, but some of them can peek at 1.2 amps on the 12v

patvdleer, 2022-11-14:
Mitsubishi MF503B(-312M), the DC power requirements show:

```
+5v   +-5% 0.5A typ 0.7A
+12v +-5% 0.6A typ 1.0A max
```

https://www.minuszerodegrees.net/manuals/Mitsubishi/Mitsubishi%20-%20MF501B,%20MF503B,%20MF504B%20-%20Specifications%20and%20Application%20Notes.pdf 

page 2 near the bottom

RobSmithDev, 2022-11-14:
I suggest you try to get the disk spinning without the arduino for the minute, sounds like theres a issue there somewhere.  The Error you get from the software is because the arduino reset its self, probably because of the power issue

patvdleer, 2022-11-14:
Different ATX PSU, same issue, I still have a Teac 14733730, that one does spin up but results in: `Error opening COM port: Last command completed successfully.` 

I've been trying a bunch of stuff but everything seems to fail.

Even the diagnostic,

```
DrawBridge aka Arduino Floppy Disk Reader/Writer V2.8.8, Copyright (C) 2017-2022 Robert Smith
Full sourcecode and documentation at https://amiga.robsmithdev.co.uk
This is free software licenced under the GNU General Public Licence V3

Running diagnostics on COM port: /dev/ttyUSB0
Attempting to open and use /dev/ttyUSB0 without CTS
Testing CTS pin
CTS OK.  Reconnecting with CTS enabled
Board is running firmware version 1.9.25  (modded for DiskChange support)

Fetching latest firmware version...
Firmware is up-to-date
Features Set: 
      o Disk Change Pin Support
      o DrawBridge PLUS
      o HD/DD Detect Enabled
      o Accurate Flux Read Mode
Testing USB->Serial transfer speed (read)
The USB->Serial adapter in use is not suitable.
Arduino UNO: The on-board adapter is not able to sustain large data transfers
Arduino Pro mini: The USB-Serial board is not able to sustain large data transfers
Arduino Nano: The USB-Serial on this board is not fast enough to sustain large data transfers

If you are using any of the devices with a CH340 converter then swap it for an FTDI one.
If you still have problems after switching, connect the Arduino using a shorter cable
and if possible directly (ie: not via a USB hub)
DIAGNOSTICS FAILED: Diagnostics failed.
```

```
[ ] DISKCHANGE 	Force DiskChange Detection Support (used if pin 12 is not connected to GND)
[X] PLUS 	Set DrawBridge Plus Mode (when Pin 4 and 8 are swapped for improved accuracy)
[ ] ALLDD 	Force All Disks to be Detected as Double Density (faster if you don't need HD support)
[ ] SLOW 	Use Slower Disk Seeking (for slow head-moving floppy drives - rare)
[ ] INDEX 	Always Index-Align Writing to Disks (not recommended unless you know what you are doing)
```